IonSystems/tiberius-robot #3: Nice symbolic link in the web app folder...

TechAUmNu, 2015-09-21:
Please put the actual files there please!!!!

camieac, 2015-09-22:
The current directory structure is to have all modules in the modules folder. Each folder starting with 'pi-' in the top level will contain only the modules required by that Raspbery Pi. I should also note here that not all modules are actually python modules, e.g the web-interface module is a python project which in its self contains modules.

The idea was to add the modules in the modules folder, and add to the pi folders as git submodules, but I have been reading that git submodules are not all that nice, so haven't bit the bullet yet. As a temporary measure I have used symbolic links.

This is less than ideal, because the whole point of this directory structure is to allow people, or devices, to check out only the parts of the repo that they need. If I were to checkout the pi-web-interface folder I would only have a broken symbolic link in the folder.

So, the current solution would be to add the modules as git submodules and document the process of a sparse checkout. I may do this today.

camieac, 2015-09-22:
Ok, with `git submodules`, you can only attach the module to HEAD or a specific revision. It doesn't allow you to clone only a folder in the repo, so won't work for linking to modules unless they were in their own repository.

camieac, 2016-01-24:
Issue closed, a decision was made to install all *compatible* modules for each installation. See https://github.com/IonSystems/tiberius-robot/blob/master/setup.py for exactly what is installed. There are no longer folders for each deployment device, as of https://github.com/IonSystems/tiberius-robot/commit/2f0a69b88b30533e0e7cd157c3c960d241f91efc.

closed 2016-01-24 by camieac