IonSystems/tiberius-robot #5: Control Architecture

camieac, 2015-09-25:
# Current Idea for implementing shared access to control (sensors and actuators):

We have a module called `control`, this has access to all sensors and actuators. We have many other modules that will need to use `control`, and may also me running concurrently. How do we allow access to control without modules interrupting each other?

A queue: We hide the actual `control` functions from the other modules and only allow access to the control queue. Because we can have sensors and motors being driven at the same time, we could have multiple queues, one for each piece of hardware that we need to manage, and enqueue from these concurrently.

What about synchronisation? Hmm, yes. Imagine we want to drive to a GPS coordinate and collect LIDAR data. So we do this:
```
from control.queues import MotorQueue
from control.queues import LIDARQueue

mq = MotorQueue()
lq = LidarQueue()

if __name__ == "__main__":
    #Cheating me presuming coordinate is 100 units forward.
    mq.add(mq.FORWARD, distance = 100)
    lq.add(lq.CAPTURE)

```

Control would then dequeue from the motor and lidar queue at the same time and they would be executed concurrently. So multiple queues on their own wont work.

Let's roll back to the idea of one queue, this allows us to do one thing at a time, by dequeuing, executing then dequeuing. What if we check what the task is before dequeuing it? This would allow tasks to remain in the correct execution order, and also allow tasks to run concurrently if they can be, by checking if an instance of the particular type of task in question is already running. This will also require locks for each type of task that can be run.

Possible code:
```
while(True):
   if (task = q.readNextTask()): #If there is a next task in the queue
        if(resourceAvailable(task['type'])):
            task = q.dequeNextTask()
            executeTaskInThread(task)
            
```
`executeTaskInThread()` would take care of the locks for that particular resource, so the while loop could still execute other tasks if the resource is available.

What I have described so far still doesn't answer the original problem, because the move and the LIDAR sense would still be done concurrently. Duh! To be continued...

# Alternative
Use ROS and all this is taken care of.

TechAUmNu, 2015-09-25:
Or you could just shove all the sensor data into the database. 

For the motors they should respond as fast as possible to any command. i.e. the most high priority command should be taken. Since it is a physical actuator safety is a critical part of any design. So for example if a stop command was sent then that should override any other command (and ideally it should have some kind of lock so that a stop has to be 'unlocked' before it can start moving again)

camieac, 2015-09-26:
I didn't consider the database. I'll try to put a schema together over the weekend. For the motors, I guess we will just have an emergency stop method that stops the motors and locks them in the off position until a special 'emergency resolved' method is called. The database could act as a state machine for this, similar to what it was before (https://github.com/IonSystems/tiberius-robot/blob/master/modules/database/newdatabase/tables.sql). 

Unfortunately for me, I don't get to implement a priority queue.

TechAUmNu, 2015-10-25:
This looks useful, might be easier to use than polyhedra. http://www.mysql.com/products/cluster/
It supports in memory databases. Also it looks like normal mysql would probably work just fine for our purposes and is probably the best thing to use anyway.

TechAUmNu, 2015-10-25:
This also looks like a nice way to run it https://www.adminer.org/