IonSystems/tiberius-robot #6: Power Management Idea

camieac, 2015-09-27:
I just thought as I was thinking about the diagnostics for Tiberius. It would be awesome if a Pi could manage power.

We could have a Pi/Arduino connected to Ethernet, this device would be connected to lots of relays/transistors to control power to each device on Tiberius. On startup, all devices would have power. The device controlling power obviously will need fixed power, otherwise it could switch itself off.

Any device connected to Ethernet will be able to disable power to other devices or even itself. This will be useful for diagnostics, power saving.

I am sure Euan will say he was thinking this already.

Here is a quick and dirty concept:
[image]

TechAUmNu, 2015-10-25:
The problem with this is then you are using loads of power on the Pi that controls the power, it really needs to be a much lower power unit that does the management, more like a microprocessor with internet or something. Unless the power the power Pi uses is negligible.

I was also thinking we could have every usb device plugged in with a relay going to the io pin on the Pi it is connected to, that way the pi using the sensor can turn it off when it isn't taking readings.

camieac, 2015-10-25:
Yeah I was thinking relays too. We could use a low power microcontroller.

TechAUmNu, 2015-10-25:
This is rather interesting, The A+ uses significantly less power than the Pi 2. https://github.com/geerlingguy/raspberry-pi-dramble/wiki/Power-Consumption

Its actually a lot less than I thought