IonSystems/tiberius-robot #15: Power Management Module

camieac, 2015-10-24:
A python module that contains power a management control code. Will contain algorithms for working out when to power down certain units. You may want to consider adding the following:
- A thread to update the database with power usage details to generate historical power data.
- Contribute a driver to the  [control](https://github.com/IonSystems/tiberius-robot/tree/master/modules/control) module for interfacing with the TI power monitoring chip.
- Decide if you want a dedicated Power Pi, or run the control algorithms on the Control Pi or Database Pi.
There is bound to be a lot more things that you can do that I can't think of at the time of writing.

TechAUmNu, 2016-02-03:
It makes sense for this module to be more about control of power components, eg the lights, power to arm, etc. This should probably be done by interfacing an arduino or other microcontroller and sending serial commands to it over usb to control power to each device using relays.

AndrewRigg, 2016-03-16:
We have looked at certain hardware that will allow voltage and current values to be read from the batteries.  This most likely requires a ~£20 chip to be bought which was added to the parts list.