IonSystems/tiberius-robot #22: I2C Errors

camieac, 2016-01-11:
I noticed an unmanageable amount of I2C errors tonight, while testing the control software. I think this is something to do with commit 9bb996b when I may have removed some important delays.

I will test on another installation at some point and see if there is still a problem.

Errors in logging:
```
2016-01-11 18:15:46,801 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:46,880 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:46,995 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:47,076 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:47,158 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:47,238 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:47,351 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:47,433 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:47,515 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:47,594 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:47,708 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:47,790 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:47,871 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:47,951 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:48,072 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:48,154 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:48,236 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:48,312 - tiberius.control.UltrasonicRangefinder - ERROR - IO error getranging 0x73
2016-01-11 18:15:48,319 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:48,433 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:48,515 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:48,597 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:48,676 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:48,790 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:48,872 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:48,954 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:49,033 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:49,148 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:49,229 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:49,311 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:49,390 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:49,504 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:49,587 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:49,668 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:49,748 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:49,862 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:49,944 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:50,025 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:50,105 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:50,219 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:50,300 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:50,382 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:50,466 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:50,580 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:50,662 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:50,743 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:50,823 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:50,937 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:51,019 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:51,100 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:51,179 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:51,294 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:51,375 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:51,457 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:51,536 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:51,650 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:51,732 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:51,813 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:51,893 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:52,007 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:52,088 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:52,170 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:52,249 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:52,363 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:52,446 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:52,527 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:52,606 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:52,721 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:52,802 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:52,883 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:52,963 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
2016-01-11 18:15:53,078 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:53,160 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:53,241 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73

2016-01-11 18:15:53,078 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:53,160 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:53,241 - tiberius.control.UltrasonicRangefinder - ERROR - I2C write error 0x73
2016-01-11 18:15:53,321 - tiberius.control.Control - DEBUG - Front Right : True ,Front Centre: True , Front Left: True
```
I am also wandering why only device 0x73 is giving an error, this could show a hardware issue.

camieac, 2016-01-11:
And it is probably worth setting a limit on the number of read/write attempts, rather than looping until succesful.

TechAUmNu, 2016-01-11:
If you are just looping until successful then that is why 0x73 is the only
one erroring. If I remember rightly doesn't i2c REQUIRE a delay?

TechAUmNu, 2016-01-11:
I would make a module that extends the smbus library, then you can put in delays easily between each transaction

camieac, 2016-01-11:
Yeah you're right about why it's only 0x73. Feel free to have a look at this with me tomorrow.

camieac, 2016-01-11:
Shouldn't need delays (it has worked fine without them previously) , after closer inspection of the commit referenced above (9bb996b), there were never any delays. If anything the changes made have increased the delays ever so slightly.

camieac, 2016-01-13:
The issue with the motors not stopping was caused by two of the motor drivers having an old firmware version (v7 rather than v15). The old firmware dealt with the direction register slightly differently, causing the weird issue. This has been fixed in Tiberius Junior (the one with all the broken axles), by replacing the old motor drivers with new ones.

We may run into this issue again if the old motor drivers need to be used on Tiberus III. But we will cross that bridge when we come to it. I have added some support functions (92d7cad46c45ffe5c5255881e654999540fea69d) to make it easy to deal with the old drivers when it comes to it.

Finding the root of the ultrasonics errors is next on the list.

closed 2016-05-13 by camieac