IonSystems/tiberius-robot #26: GPS Waypoint Navigation

camieac, 2016-01-18:
We need to be able to navigate between two GPS way points. This will firstly involve driving directly between two way points with no obstacle avoidance.

After we have a working algorithm for navigating between two points, be can add more complex elements to help with obstacle avoidance.

This issue is only for the basic waypoint navigation function, e.g.
```
def driveTo(latitude, longitude, speed):
```
Or something along similar lines.

camieac, 2016-01-20:
Dennis done some work with GPS in here that you may find useful, although some of it is impossible to understand. https://github.com/IonSystems/tiberius-robot/blob/master/tiberius/autonomy/mission_control.py

camieac, 2016-02-02:
As of 1a1ec0314ec6cd550e34dfbee71a2f3e3f2cfd1e the GPS point-to-point algorithm looks mostly done, although we haven't had a chance to test it yet. This issue will stay open until we are happy with the algorithms and the gps-algorithms branch is commited into master with suitable tests.

TechAUmNu, 2016-02-03:
The point to point algorithm was tested today and a number of problems were found. These have now been fixed in 97108e64f3334021a7b19cf95a0f9df7b614dd9a

closed 2016-02-09 by camieac